xioTechnologies/Fusion #138: Roll and pitch relative to device, yaw relative to earth

copumpkin, 2023-12-24:
Apologies if I’m misunderstanding here, but as I understand it Fusion’s 3 orientation reference frames are relative to earth in all 3 coordinates.

I’m trying to derive roll and pitch relative to my IMU, and earth-relative yaw, so effectively I’d get a “compass” reading from yaw, but I can easily determine if the IMU has tilted side to side or front to back and isolate sideways tilt from front/back tilt.

To achieve this, it seems like I could use yaw straight out of Fusion but to get my device-relative pitch and roll, I’d have to adjust the other two Euler angles by my yaw using some trig. Is that correct or am I misunderstanding the convention?

Also, if the above is roughly correct, which yaw values correspond to north/south/east/west?

Thanks!

xioTechnologies, 2023-12-28:
Fusion uses the [ZYX Euler angles](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionMath.h#L466) sequence with an [axes convention](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionConvention.h#L20) selected in the [settings](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.h#L25), either NWD, ENU, or NED.

It sounds like your enquiry is not about Fusion and you just need clarification on what Euler angles actually are.  I suggest you search online.  There are [countless YouTube videos](https://www.youtube.com/results?search_query=euler+angles) that do a far better job of explaining Euler angles with animations than I could do in written comments.

copumpkin, 2023-12-28:
You're right! I think I was having some issues with my inputs into Fusion which was then producing counterintuitive results which led me to think I was getting the wrong frame of reference. After some tinkering I think it's behaving more in line with how I'd expect now, thanks!

closed 2023-12-28 by copumpkin