xioTechnologies/Fusion #150: Measurement convention for sensors other than x-imu

maker-ATOM, 2024-02-16:
For to use Fusion algorithm for motion tracking of objects like demostrated in [Gait-tracking](https://github.com/xioTechnologies/Gait-Tracking) with different sensors available like BNO055, it is necessary for the sensor to include gravitational measurements within Accelerometer.

For example if the sensor is lying flat on a table surface with z-axis pointing upwards should the sensor measure (0.0, 0.0, 0.0) g or (0.0, 0.0, 1.0) g.

xioTechnologies, 2024-02-19:
I do not understand your comment.  If you are asking a question then please phrase your comment as a question.

maker-ATOM, 2024-02-23:
If I have to use Fusion Algorithm with sensors other than x-imu what conventions do I need to follow?

xioTechnologies, 2024-02-23:
Fusion is a general purpose library.  There is nothing about Fusion that ties it to the x-IMU or any other product.

closed 2024-04-22 by xioTechnologies