xioTechnologies/Fusion #158: Q: Influence of motion and external acceleration

ib00, 2024-04-17:
How does motion and external acceleration effect fusion algorithm?

For example, if I am on a bike pedaling, and the IMU sensor is on my leg, how does this bike motion/acceleration
affect the algorithm? Do we need to change anything?

xioTechnologies, 2024-04-17:
I suggest you monitor the _acceleration recovery flag_ as an indication of algorithm health.  If the flag is never (or rarely) set then the motion is not an issue.

ib00, 2024-04-25:
Thanks! What about extreme acceleration of the sensor? I have an application (think of it as running-like application) where sensor accelerates rapidly and then it comes to a complete stop when it hits the ground.

x-IMU3 sensor has a range up to 24g, but I have measured it to be much much higher.

xioTechnologies, 2024-04-25:
As I said, I suggest you monitor the flag.  If it is never set then the acceleration (extreme or otherwise) is not an issue.  If you need to measure accelerations >24g then you should use the 200g accelerometer.

closed 2024-08-15 by xioTechnologies