xioTechnologies/Fusion #159: Very noisy Roll Pitch  Yaw data

YunHB, 2024-04-19:
Hello. Thank you for those nice libraries.

I'm trying to use those libraries for MARG sensor fusion, and when I logged those data I got data with huge noise.
I am thinking of calibration of sensors and introduction of low-pass filters for sensor data as a solution to the noise.

May I get some hints with your opinion?

I used the c headers and sources.

those pictures are compared shots with commercial IMU module.
[image]
[image]
[image]

xioTechnologies, 2024-04-19:
It looks like your Euler angles are garbage, not noisy.  I suggest you format your sensor data into a [CSV file](https://github.com/xioTechnologies/Fusion/blob/main/Python/sensor_data.csv) to be processed by [simple_example.py](https://github.com/xioTechnologies/Fusion/blob/main/Python/simple_example.py).  This will plot gyroscope, accelerometer, and Euler angles side-by-side as shown below.  This plot may offer some insight as to what is going on.

[image]

YunHB, 2024-04-22:
It looks like garbage, but with same imu, same sensors, other algorithm got very accurate euler angle data.
And the difference I found between those two algorithm is existence of LPF on sensor data.
I will try your suggestion, but I think it won't make this better.

YunHB, 2024-04-23:
OH that was my misunderstanding, sorry. I applied LPF and HPF on sensor data, and I got a better plot. Maybe the sensors have too much noise or I treated them in unstable condition. 
[image]
[image]
[image]

xioTechnologies, 2024-04-23:
If you are not going to engage with this support then please close the ticket.

closed 2024-04-24 by YunHB