xioTechnologies/Fusion #161: yaw value changes while rotating on only roll or pitch axis.

YunHB, 2024-04-25:
Hi. Thank you for your algorithm and kind answers on other issues.

I have some problem on yaw value, especially for 2 phenomenon.

First, yaw value moves when I DON"T rotate on yaw axis.
I simply rotated my imu on roll or pitch axis, yaw value changed.

Second, when I rotate on yaw axis, yaw values are not changing as uniform degree.
When the rotation plane is divided into four quadrants, it changes like 80, 80, 80, 120 degree on each quadrant.

I'm implementing experiences changing SAMPLE_RATE, gyroscopeRange, recoveryTriggerPeriod, etc.

Do you have any suggestions on why this is happening?

I'll attach some plots if they would help.

xioTechnologies, 2024-04-25:
I cannot offer any suggestions when you provide so little information.  I suggest you rewrite your post so that others may be able to help.  You would need to provide a description of your setup and clearly explain what is not working.  This is usually best expressed as a description of what you expect to happen juxtaposed with what is actually happening.  In this case, I expect annotated graphs would be useful.

YunHB, 2024-04-26:
[image]
This is ahrs setting for my code, and 
[image]
this is graph with Blue - roll, Red - pitch, green - yaw.
Sample index 100~600 : IMU only rotated on roll direction.
Sample index 600~1000 : IMU only rotated on yaw direction.
Sample index 1080~1400 : IMU only rotated on pitch direction.

xioTechnologies, 2024-04-26:
Your posts are not written in a way that could allow others to help.  I am unable to support you.

closed 2024-06-14 by xioTechnologies