xioTechnologies/Fusion #166: 5 seconds delay in real-time operation (python)

paulasimoUPC, 2024-05-06:
Hi, 

Thank you for sharing your work with the community. I am reading the data from a sensor in real-time through serial connection and processing it in real-time. I have noticed a five seconds delay on the calculation of the orientation using python. I suspect the error comes from the time that the library needs to process the data. Is it possible to correct this?

Please let me know if you need more information to be able to answer my question. 

Thank you for your time.

xioTechnologies, 2024-05-06:
No, your suggestion is not correct.  The algorithm processing time has not been profiled for Python but I expect it will be in the order of microseconds.

paulasimoUPC, 2024-05-09:
Hi,

Thank you very much for your fast response. After many trials, I have detected that, for using the library in Python in real-time, the parameters attached below work perfectly for my sensor (XSens MTi-28A33G35).

ahrs.settings = imufusion.Settings(imufusion.CONVENTION_ENU,  # convention
                                   10,  # gain
                                   1200,  # gyroscope range
                                   0,  # acceleration rejection
                                   0,  # magnetic rejection
                                   0)  # recovery trigger period

ahrs.update(gyroscope, accelerometer, magnetometer, 1/100)
                                   
Setting the recovery trigger period to 0, which also turns the acceleration and magnetic rejection to 0, helped me achieve a real-time time response (the value changed fast after moving the sensor). However, after moving the sensor from one orientation to another one, it was still taking some seconds to converge to the new value. Setting the gain to 10 helped me converge to the new value in a matter of ms. 

I am still not fully understanding why these parameters are the adequate ones for my sensor. I would expect that relying more on the gyroscope than on the others sensors would cause an accumulative error over time due to the integration of the sensor noise, however I am not seeing this behavior. Why does a higher gain improve the performance? Also, are the acceleration and magnetic rejection very computationally costly (that might justify the delay while using them)?

Thank you very much for your help and guidance.

closed 2024-06-14 by paulasimoUPC