xioTechnologies/Fusion #167: Need some help setting up the algorithm.

brightproject, 2024-05-19:
Thanks to the team @xioTechnologies
I'm getting raw gyroscope angular velocity and accelerometer acceleration data from the `FlightGear` flight simulator.
I managed to compile a simple example for `stm32f411`, for compilation I used the Arduino platform.
Raw data sampling
*
performed at about 30-40 Hz.
[image]

Set the value
`#define SAMPLE_PERIOD (0.04f) // 40Hz.`
Perhaps I think this meaning is wrong?
Example code I'm using.
```
#include <Fusion.h>
#include <stdbool.h>
#include <stdio.h>

// #define PRINTF

#define SAMPLE_PERIOD (0.04f) // 40Hz.

#define SERIAL_BAUD 115200

FusionAhrs ahrs;

void setup() {
	
	Serial.begin(SERIAL_BAUD);// PA9, PA10
	
	FusionAhrsInitialise(&ahrs);

}

void loop() { // this loop should repeat each time new gyroscope data is available
        const FusionVector gyroscope = {0.2f, 1.3f, 0.0f}; // replace this with actual gyroscope data in degrees/s
        const FusionVector accelerometer = {0.0f, 0.0f, 1.0f}; // replace this with actual accelerometer data in g

        FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, SAMPLE_PERIOD);

        const FusionEuler euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
		
		#ifdef PRINTF

        printf("Roll %0.1f, Pitch %0.1f, Yaw %0.1f\n", euler.angle.roll, euler.angle.pitch, euler.angle.yaw);
		
		#else
		
		Serial.print(" \tEiler Roll: ");
		Serial.print(euler.angle.roll);
		Serial.print(" \tEiler Pitch: ");
		Serial.print(euler.angle.pitch);
		Serial.print(" \tEiler Yaw: ");
		Serial.println(euler.angle.yaw);
	  
		#endif
}
```
I tried different parameters [from](https://github.com/xioTechnologies/Fusion/blob/e7d2b41e6506fa9c85492b91becf003262f14977/Fusion/FusionAhrs.c#L64), but did not notice any significant differences.

```c
void FusionAhrsInitialise(FusionAhrs *const ahrs) {
    const FusionAhrsSettings settings = {
            .convention = FusionConventionNwu,
            // .convention = FusionConventionEnu,
            // .convention = FusionConventionNed,
            .gain = 0.5f,
            // .gain = 0.01f,
            .gyroscopeRange = 0.0f,
            // .gyroscopeRange = 250.0f,
            // .gyroscopeRange = 500.0f,
[... 65 lines cut ...]
> gyro_x: -0.002700 gyro_y: 0.029271 gyro_z: 0.046256 accel_x: 1.119500 accel_y: -0.148700 accel_z: -11.522500 roll: 32.421532 pitch: 4.330776 yaw: 39.820187
> gyro_x: -0.002807 gyro_y: 0.029203 gyro_z: 0.046233 accel_x: 1.121500 accel_y: -0.148300 accel_z: -11.522900 roll: 32.347038 pitch: 4.335338 yaw: 39.969360
> gyro_x: -0.002933 gyro_y: 0.029119 gyro_z: 0.046203 accel_x: 1.123900 accel_y: -0.147600 accel_z: -11.527900 roll: 32.272358 pitch: 4.339952 yaw: 40.118343
> gyro_x: -0.003099 gyro_y: 0.028999 gyro_z: 0.046160 accel_x: 1.127300 accel_y: -0.146700 accel_z: -11.529900 roll: 32.197365 pitch: 4.344566 yaw: 40.267056
> gyro_x: -0.003210 gyro_y: 0.028909 gyro_z: 0.046129 accel_x: 1.129700 accel_y: -0.146100 accel_z: -11.532300 roll: 32.122211 pitch: 4.349224 yaw: 40.415554
> gyro_x: -0.003274 gyro_y: 0.028853 gyro_z: 0.046110 accel_x: 1.131000 accel_y: -0.145700 accel_z: -11.533300 roll: 32.047043 pitch: 4.353974 yaw: 40.563931
> gyro_x: -0.003355 gyro_y: 0.028776 gyro_z: 0.046085 accel_x: 1.132800 accel_y: -0.145200 accel_z: -11.533200 roll: 31.971813 pitch: 4.358785 yaw: 40.712143
> gyro_x: -0.003414 gyro_y: 0.028721 gyro_z: 0.046065 accel_x: 1.134000 accel_y: -0.144800 accel_z: -11.540500 roll: 31.896582 pitch: 4.363681 yaw: 40.860233
> gyro_x: -0.003490 gyro_y: 0.028672 gyro_z: 0.046039 accel_x: 1.135700 accel_y: -0.144300 accel_z: -11.552700 roll: 31.821310 pitch: 4.368689 yaw: 41.008205
> gyro_x: -0.003564 gyro_y: 0.028615 gyro_z: 0.046013 accel_x: 1.137300 accel_y: -0.143700 accel_z: -11.543100 roll: 31.745993 pitch: 4.373813 yaw: 41.156025
> gyro_x: -0.003634 gyro_y: 0.028536 gyro_z: 0.045987 accel_x: 1.138900 accel_y: -0.143200 accel_z: -11.537600 roll: 31.670635 pitch: 4.378995 yaw: 41.303661

[image]
I succeeded to test a simple implementation of the `xioTechnologies` fusion sensor.
https://www.youtube.com/watch?v=cWlsb3izkYQ
`GUI` on [Processing](https://processing.org/) from [here](https://github.com/VoxMi/MPU-9150) with minor modifications.
I deliberately do not try to use an advanced example because I want to practice the basic settings of the algorithm with a simple example.
How can I tell if the algorithm was initialized based on `accelerometer` and `gyroscope` data?
How to correctly use algorithm flags and how to activate them?

> https://github.com/xioTechnologies/Fusion?tab=readme-ov-file#algorithm-flags

Perhaps this behavior of the roll angle during a long turn is typical and normal for an algorithm that uses only a gyroscope and an accelerometer as input data?
Can you please tell me how to configure the algorithm to work acceptable with the data obtained from the flight simulation on `FlightGear`.
Or do you recommend going straight to the advanced example and already working on its settings?

xioTechnologies, 2024-05-19:
I'm afraid that there is just too much in the post for me to effectively answer.  I suggest you start from the [simple example](https://github.com/xioTechnologies/Fusion/blob/main/Python/simple_example.py) which does not require any configuration or setup, and then post your questions or issues so that we deal with them one at a time.

brightproject, 2024-05-19:
> I'm afraid that there is just too much in the post for me to effectively answer. I suggest you start from the [simple [...]

Thank you for your answer, I really appreciate your support and nothing bad will happen if you can’t help me.
Using a simple example, in the post above I slightly changed the output data, if you look at the graphical plotting of the orientation angles obtained from the simulator and calculated by the algorithm.

xioTechnologies, 2024-05-19:
As I said, I cannot offer a useful response to your original post.  I suggested a way we could proceed but if that does not work for you then I probably won't be able to help.

xioTechnologies, 2024-05-20:
The CSV columns should be: `Time (s),Gyroscope X (deg/s),Gyroscope Y (deg/s),Gyroscope Z (deg/s),Accelerometer X (g),Accelerometer Y (g),Accelerometer Z (g)`.  Your gyroscope and accelerometer columns look ok.

The difference between the first two time values in your CSV file corresponds to a sample rate of 26.36 Hz.  [simple_example.py](https://github.com/xioTechnologies/Fusion/blob/main/Python/simple_example.py) includes the expression `1 / 100` on [line 37](https://github.com/xioTechnologies/Fusion/blob/main/Python/simple_example.py#L37C73-L37C80) for a sample rate of 100 Hz.  You need to change this to `1 / 26.36` to match your data.

I do not understand the sentence, "_The number of samples is still 28 Hz_".

**I wrote this reply before you edited your post.  You have since added allot more your post, confusing the discussion.  I can't help you if this is how you are going to communicate.**

brightproject, 2024-05-20:
> I do not understand the sentence, "_The number of samples is still 28 Hz_".

This means that I made the first data sample and calculated that I get 28 samples/rows per second, and concluded that the sampling rate is 28 Hz.
After that, I made a second data sample, and in this I received 35 samples/rows of data per second.

> The difference between the first two time values in your CSV file corresponds to a sample rate of 26.36 Hz. [...]

Thanks, I changed the divisor number and got this result.
[image]
I play with the parameters of the algorithm, not yet understanding its operation well enough.
But I couldn’t get it to hold the bank angle in a turn for long enough.
And the main question here is - the algorithm does not allow you to compensate for centripetal acceleration during rolls and turns, or do I need to study the operation of the algorithm better and adjust it?

xioTechnologies, 2024-05-20:
Your questions about configuration, features, and expected behaviour are all moot until you can get the basics working.  It sounds like you are not sure that your sample rate is, or even if it is fixed.

Before you do anything else, you need to determine your sample ate and/or the timestamp of measurements.

brightproject, 2024-05-22:
> Your questions about configuration, features, and expected behaviour are all moot until you can get the basics working. It sounds like you are not sure that your sample rate is, or even if it is fixed.
> 
> Before you do anything else, you need to determine your sample rate and/or the timestamps of measurements.

Unfortunately, after writing one question, I continue to search for a solution to the problem, and certainly additional questions arise, and I update/edit/supplement the original request, and can’t do anything about it, please understand and forgive me.
Regarding the sample rate issue, because... I use a flight simulator, and for objective reasons I cannot get a sampling frequency from the simulator greater than 40 Hz, then I propose to assume that my samples are at this frequency.
Collected data in file

> [sensor_data_v7.csv](https://github.com/xioTechnologies/Fusion/files/15401808/sensor_data_v7.csv)

Running a `simple example` shows that the orientation angles are not stabilized and the roll quickly decreases (this is not even a roll, but a sawtooth movement), although the roll angles are long in nature
[image]

`Advanced example` used with the following settings
```
ahrs.settings = imufusion.Settings(imufusion.CONVENTION_NWU,  # convention
# ahrs.settings = imufusion.Settings(imufusion.CONVENTION_ENU,  # convention
                                   # 0.01,  # gain
                                   # 0.05,  # gain
                                   # 0.6,  # gain
                                   0.008,  # gain
                                   # 0.00000,  # gain
                                   0,  # gyroscope range
                                   # 2000,  # gyroscope range
                                   # 250,  # gyroscope range
                                   1,  # acceleration rejection
                                   # 0,  # acceleration rejection
                                   1,  # magnetic rejection
                                   # 0,  # magnetic rejection
                                   # 5 * sample_rate)  # recovery trigger period = 5 seconds
                                   1)  # recovery trigger period = 5 seconds
```
The result is as follows
[image]

It can be seen that the bank angle became longer, however, immediately after the right bank, the return to straight flight was not at 0 degrees.
And then I noticed some strange things if you specify the parameter
`0.00000, # gain`
Then the picture will be as follows
[image]

It turns out with zero sensitivity, roll angles almost always return to zero degrees, although in fact the lines just narrowed a little along the `Y axis` and jumped up a little.
I also don’t quite understand the following - the plane in the simulator rotated in the bank angle by `+-30 degrees`.
But the `simple` and `advanced` example, for unknown reasons, show roll angles of `5 degrees` - could you please comment on this?
I already looked at the example code, I thought maybe the scaling was wrong somewhere ... but with the test data `sensor_data.csv` all the angles are normal.
My python self-code shows excellent results from the collected data
[image]

The parameters of the `C++ code` algorithm are as follows:
  ```
const FusionAhrsSettings settings = {
  // info about convention
  // https://github.com/xioTechnologies/Fusion/issues/117#issuecomment-1677998150
          .convention = FusionConventionNwu,
          // .convention = FusionConventionEnu,
          // .convention = FusionConventionNed,
          // .gain = 1.5f,
          // .gain = 0.6f,
          .gain = 0.008f,
          // .gain = 0.0079f,
          .gyroscopeRange = 0.0f, /* replace this with actual gyroscope range in degrees/s */
          // .gyroscopeRange = 250.0f,
          // .gyroscopeRange = 500.0f,
          // .gyroscopeRange = 1000.0f,
          // .gyroscopeRange = 2000.0f,
          // .accelerationRejection = 5.0f,
          .accelerationRejection = 0.0f,
          // .accelerationRejection = 0.6f,
          // .magneticRejection = 90.0f,
          .magneticRejection = 0.0f,
[... 1 lines cut ...]
          // .recoveryTriggerPeriod = 0.14,
          // .recoveryTriggerPeriod = 0.142,
          // .recoveryTriggerPeriod = 5 * SAMPLE_PERIOD, /* 5 seconds */
          // .recoveryTriggerPeriod = 37500 * SAMPLE_PERIOD, /* 5 seconds */
  };
```
These are quite acceptable results for me and there are plans to improve them using a magnetometer.
I verified that the algorithm could maintain roll and pitch angles despite prolonged tilts, ascents, and descents.
I was able to tune the algorithm only by selecting the `SAMPLE_PERIOD` and `.gain` coefficients
Based on the results of searching for coefficients, I settled on the following
```
SAMPLE_PERIOD (0.0342f)
.gain = 0.008f
```
And again, I noticed some strangeness, or rather a pattern, which in no way relates to the fusion algorithm.
The pattern applies to my code, which is executed on `STM32F411` and it is as follows - when I execute the code without collecting data samples, then with the SAMPLE_PERIOD coefficient indicated above, the fusion algorithm is not bad, it maintains the roll and pitch angles without errors for `30-60 seconds`.
But if I want to collect data samples, then I introduce an additional load into the microcontroller code, and the microcontroller executes the program, including
`FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, SAMPLE_PERIOD);`
[... cut ...]
I also thought that this notorious `SAMPLE_PERIOD` coefficient is similar to the `sample_rate` coefficient from an `extended example` in Python.
But it turned out that the `sample_rate` coefficient cannot even be given a fractional value.
I probably still don’t fully understand the magic of the settings of the fusion algorithm, but I understand that the operation of the algorithm greatly depends on the quantity, i.e. frequency of data from the gyroscope and accelerometer, as well as selection of coefficients in the code section:
`imufusion.Settings`
I hope another long text did not confuse you.
Thank you.

xioTechnologies, 2024-05-22:
I have asked that we tackle issues one at a time, and I identified the unknown sample rate as being the critical issue that must be solved before anything else.  I can't support you unless you engage with what I am saying.

brightproject, 2024-05-22:
> I have asked that we tackle issues one at a time, and I identified the unknown sample rate as being the critical issue that must be solved before anything else. I can't support you unless you engage with what I am saying.

My SAMPLE_PERIOD is 0.0340 seconds, then:
```
Frequency = 1 / 0.0340
Frequency ≈ 29.4118 Hz
```
Thus, a `SAMPLE_PERIOD` of `0.0340` seconds corresponds to a frequency of approximately `30 Hz`.
​

xioTechnologies, 2024-05-22:
The first step is for you to run [simple_example.py](https://github.com/xioTechnologies/Fusion/blob/main/Python/simple_example.py) with valid data.  This means using the correct sample rate and timestamps.  I am not confident that the latest sample rate you have reported is correct because you have reported numerous other conflicting values:
- "_The number of samples is still 28 Hz_"
- [sensor_data_v4.csv](https://github.com/xioTechnologies/Fusion/files/15375816/sensor_data_v4.csv) - The timestamps in this file correspond to 26.36 Hz.
- "_Here's a 35 Hz sample_"
- "_from the simulator greater than 40 Hz, then I propose to assume that my samples are at this frequency_"
- [sensor_data_v7.csv](https://github.com/xioTechnologies/Fusion/files/15401808/sensor_data_v7.csv) - The timestamps in this file correspond to 10 Hz.
- "_Frequency ≈ 29.4118 Hz_"
- "_frequency of approximately 30 Hz_"

I was expecting you to confirm the actual sample rate, verified in such a way that previously reported values could be discounted.

xioTechnologies, 2024-05-22:
I want to help you but you refuse to engage with the process.  I am leaving the discussion.  Good luck.

closed 2024-05-22 by xioTechnologies

brightproject, 2024-06-19:
> I want to help you but you refuse to engage with the process. I am leaving the discussion. Good luc

@xioTechnologies If may, I ask a question?
I managed to collect data from the flight, I filtered this data and put it into an advanced filter example.
I first rotated the `X` and `Y` axes of the gyroscope, because The reference system of the aircraft is` NED`, but for the example you need `ENU`.
The filtered data file is below.
[real__flight_data_filtered.csv](https://github.com/user-attachments/files/15905934/real__flight_data_filtered.csv)

The filter parameters are as follows.
```
ahrs.settings = imufusion.Settings(imufusion.CONVENTION_ENU,  # convention
                                   0.7,  # gain
                                   0.8,  # gyroscope range
                                   0.8,  # acceleration rejection
                                   10.5,  # magnetic rejection
                                   # 5 * sample_rate)  # recovery trigger period = 5 seconds
                                   600)  # recovery trigger period = 5 seconds
```
Result
[image]
It's impossible to figure this out without your help.
Thank you.