xioTechnologies/Fusion #179: Anomaly quaternion when applying realtime calculating by sensor data from UDP connection

GongTen, 2024-09-03:
Hello! 

I am experiencing an issue that appears to be related to precision overflow when calculating quaternions from real-time sensor data. I am using the IMU Utility software on the iOS platform to push sensor data via UDP and compute quaternions for simulating object rotations on the screen.

However, when using the following code, I receive quaternion values that seem abnormal due to a possible precision issue: [ 4.6025425e-36  1.4012985e-45 -6.6520590e-01  1.9545767e-02 ].

“”“
ahrs = imufusion.Ahrs()
ahrs.settings = imufusion.Settings(imufusion.CONVENTION_ENU,  # convention
                                    0.5,  # gain
                                    2000,  # gyroscope range
                                    10,  # acceleration rejection
                                    10,  # magnetic rejection
                                    5 * 50)  # recovery trigger period = 5 seconds
ahrs.update(gyro, acc, mag, 0.02)
quaternion = ahrs.quaternion.wxyz
”“”

Interestingly, when I write the real-time data to a file and process it using the same code, the calculated quaternions appear normal.

I would appreciate any guidance or suggestions on resolving this issue. Could it be related to the way real-time data is handled versus file data?

Thank you for your assistance.

PS：
This is the realtime data and the quaternions calculated in realtime:
[data5.csv](https://github.com/user-attachments/files/16843170/data5.csv)
This is the file contains the quaternions calculated by the same sensor data from the file above:
[data6.csv](https://github.com/user-attachments/files/16843176/data6.csv)

xioTechnologies, 2024-09-03:
My understanding from your description is that you have 4 floats and when you write those floats to file, all is ok, but when you send through some communication channel, the floats get corrupted.  It sounds like you need to debug your communication channel.

GongTen, 2024-09-03:
Perhaps my description is unclear. 

**Issue Description：**
I am able to obtain certain sensor data in real-time at a frequency of 50 Hz. For processing this data, I have two methods:
1.	Real-time processing: I use the code ahrs.update(gyro, acc, mag, 0.02) to convert the sensor data to quaternions in real-time.
2.	Post-processing: I write the sensor data to a file and, after data collection is complete, use the same code to convert the sensor data at each timestamp to quaternions.

Note: The sensor data used for calculations in both methods can be considered identical.

**Problem:**
For method 1 (real-time processing), the quaternions I obtain are abnormal, as shown in the file data5.csv. An example quaternion might look like this: [ 4.6025425e-36 1.4012985e-45 -6.6520590e-01 1.9545767e-02 ].

However, for method 2 (post-processing), using the same code with the same sensor data but in a non-real-time manner, the quaternions are normal, as shown in data6.csv.

**Question**
I’m curious about the possible causes of this issue. Do you have any experience that could help explain under what circumstances the ahrs.update(gyro, acc, mag, 0.02) function, followed by retrieving the quaternion with quaternion = ahrs.quaternion.wxyz, might produce such abnormal results, like [ 4.6025425e-36 1.4012985e-45 -6.6520590e-01 1.9545767e-02 ]?

**Summary**
In summary, the issue I am encountering is that the quaternions produced using the same data and code differ when processed in real-time versus non-real-time.

xioTechnologies, 2024-09-03:
You seem to be suggesting that the `update` function behaves differently for identical inputs depending where the inputs came from.  If the inputs are identical then the outputs will be identical.  If you are seeing different outputs then you must be providing different inputs.

GongTen, 2024-09-03:
Thank you for your response. It seems like I know where the problem is.
When I use the quaternion, if I use `ahrs.quaternion.wxyz`, I will get `array([1.0169907e+30, 1.4012985e-45, 1.0969947e-37, 1.6815582e-44],dtype=float32)` which is anomaly quaternion;

But if I use `np.array(ahrs.quaternion.wxyz)` I will get `array([ 0.96273106, -0.17886156,  0.06692591,  0.19151586], dtype=float32)` which is correct.

Although I’m still not clear about the cause of the issue, I could continue my research. Thank you!

xioTechnologies, 2024-09-10:
Thank you for sharing this.  There may be an issue with [`quaternion_get_wxyz`](https://github.com/xioTechnologies/Fusion/blob/main/Python/Python-C-API/Quaternion.h#L41C18-L41C37).  We are investigating.

xioTechnologies, 2024-09-27:
This bug has been fixed in [v1.2.7](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.7).

closed 2024-09-27 by xioTechnologies