xioTechnologies/Fusion #180: Overshooting and slow setting value

Gabriele-Cano, 2024-09-03:
Hi, I'm having some trouble getting relaiable and precise values from an ICM42688 using the Fusion algorithm on an ESP32-C3 (custom PCB). I implemeted Fusion algorithm following the Simple example, the pitch value is then printed on serial port to PC where a Python program shows a graph of that (white line).

My PCB is attached to a slab that can oscillate from one position to another. During the test i was moving the slab in a way that should output a "square wave" of the pitch value (in red the output i was expecting).
As you can see in the picture there is quite a bit of overshoot and the pitch value takes some seconds to adjust to the correct value. 

I believe this as something to do with the way i move the slab, infact I slam it from one side to the other quite aggressively (this is to simulate the final use).

Can someone help me solving this problem? 
Thank you!

[image]

```cpp
#include <Arduino.h>
#include "ICM42688.h"
#include <math.h>
#include <Fusion.h>
#include <stdbool.h>
#include <stdio.h>

// ICM42688 object with the sensor on I2C bus 0 with address 0x68
ICM42688 IMU(Wire, 0x68);

#define SAMPLE_PERIOD (0.07f) // replace this with actual sample period

FusionAhrs ahrs;

void setup() {
  Serial.begin(115200);

  // Configure I2C pins
  Wire.begin(5, 6);

  // Start communication with IMU
  delay(500);
  int status = IMU.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {
      Serial.println("IMU initialization unsuccessful");
      delay(100);
    }
  }

  FusionAhrsInitialise(&ahrs);
}

void loop() {
  // Read the sensor data
  IMU.getAGT();

  // Get accelerometer and gyroscope data
  float ax = IMU.accX();
  float ay = IMU.accY();
  float az = IMU.accZ();
  float gx = IMU.gyrX();// * M_PI / 180.0f; // Convert to radians
  float gy = IMU.gyrY();// * M_PI / 180.0f;
  float gz = IMU.gyrZ();// * M_PI / 180.0f;

  const FusionVector gyroscope = {gx, gy, gz}; // replace this with actual gyroscope data in degrees/s
  const FusionVector accelerometer = {ax, ay, az}; // replace this with actual accelerometer data in g

  FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, SAMPLE_PERIOD);
  const FusionEuler euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
  if (Serial.available() > 0) {
    Serial.printf("%0.2f\n", euler.angle.pitch, euler.angle.yaw);
  }
  delay(70);
}

```

xioTechnologies, 2024-09-05:
I expect a sample period of 70 ms is far too slow for motion you have described.

Gabriele-Cano, 2024-09-05:
I changed the code to have a delay of about 2ms but the output is worse because it takes even more time to reach the expected value. I tried even lower at 125us but still didn't work. (I set the ICM42688 to 8kHz ODR)

```cpp
void loop() {

  if (esp_timer_get_time() - prevMicros >= 2000) {
    prevMicros = esp_timer_get_time();
    IMU.getAGT();
    float ax = IMU.accX();
    float ay = IMU.accY();
    float az = IMU.accZ();
    float gx = IMU.gyrX();
    float gy = IMU.gyrY();
    float gz = IMU.gyrZ();

    const FusionVector gyroscope = {gx, gy, gz}; // replace this with actual gyroscope data in degrees/s
    const FusionVector accelerometer = {ax, ay, az}; // replace this with actual accelerometer data in g

    FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, (esp_timer_get_time() - prevMicros)/1000000.0f);
    const FusionEuler euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
    pitch = euler.angle.pitch;
  }

}
```
[image]

xioTechnologies, 2024-09-06:
The rate at which you read data needs to match the Output Data Rate (ODR).  I suggest you start by configuring the ICM42688 ODR for 100 Hz.

You should only read data from the ICM42688 if new data is available.  Does your `IMU` object provide a method for this?  it is usually called something like _data ready_.

Your handling of time measurement needs attention.  Here's what your code should look like:

```cpp
void loop() {
  if (IMU.dataReady()) { // does the IMU class provide something like dataReady?
    IMU.getAGT();
    float ax = IMU.accX();
    float ay = IMU.accY();
    float az = IMU.accZ();
    float gx = IMU.gyrX();
    float gy = IMU.gyrY();
    float gz = IMU.gyrZ();

    const FusionVector gyroscope = {gx, gy, gz}; // replace this with actual gyroscope data in degrees/s
    const FusionVector accelerometer = {ax, ay, az}; // replace this with actual accelerometer data in g

    const uint64_t micros = esp_timer_get_time(); // only read time ONCE and then reuse the variable wherever needed
    static uint64_t prevMicros; // static means that the value will 'be remembered' each loop
    const float deltaTime = (float) (micros - prevMicros ) / 1000000.0f; // reduce the size of the uint64_t value BEFORE casting to float
    prevMicros = micros; 

    FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, deltaTime);
    const FusionEuler euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
    pitch = euler.angle.pitch;
  }

}
```

Gabriele-Cano, 2024-09-06:
The library unfortunately didn't have any way to check if new data was ready, so i had to modify the PCB in order to connect the INTerrupt pin of the ICM42688 to the ESP (because I made a mistake and connected it to a pin that I can't use) and I modified the code accordingly. 

Now it works much much better, there is a slight overshoot but way less than before and it takes less time to reach the expected value. Do you think that by increasing the ODR (and decreasing the delay in the loop) it will become even better? 

```cpp

FusionAhrs ahrs;
volatile bool IMUready = false;

void setup() {
  int status = IMU.begin();
  IMU.setGyroODR(ICM42688::odr100);
  IMU.setAccelODR(ICM42688::odr100);
  IMU.enableDataReadyInterrupt();

  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {
      Serial.println("IMU initialization unsuccessful");
      delay(100);
    }
  }

  FusionAhrsInitialise(&ahrs);

  attachInterrupt(digitalPinToInterrupt(10), [] {
    IMUready = true;
  }, RISING);

}

void loop() {

  if (IMUready) {
    IMU.getAGT();
    float ax = IMU.accX();
    float ay = IMU.accY();
    float az = IMU.accZ();
    float gx = IMU.gyrX();
    float gy = IMU.gyrY();
    float gz = IMU.gyrZ();

    IMUready = false;
    const FusionVector gyroscope = {gx, gy, gz}; // replace this with actual gyroscope data in degrees/s
    const FusionVector accelerometer = {ax, ay, az}; // replace this with actual accelerometer data in g

    const uint64_t micros = esp_timer_get_time(); // only read time ONCE and then reuse the variable wherever needed
    static uint64_t prevMicros; // static means that the value will 'be remembered' each loop
    const float deltaTime = (float) (micros - prevMicros ) / 1000000.0f; // reduce the size of the uint64_t value BEFORE casting to float
    prevMicros = micros; 

    FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, deltaTime);
    const FusionEuler euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
    pitch = euler.angle.pitch;
  } 

  delay(5);

}
```

[image]

Gabriele-Cano, 2024-09-07:
I increased the ODR to 4kHz (not sure if it's too much or not) and it seems to have improven the results even more.
[image]

xioTechnologies, 2024-09-07:
There should not be a delay in the loop.  This will only introduce errors to the calculation of `deltaTime`.

Gabriele-Cano, 2024-09-07:
Yes, I already deleted that line, I forgot to mention that.

closed 2024-09-21 by xioTechnologies