xioTechnologies/Fusion #182: How to get the Gravity data

GongTen, 2024-09-09:
Hello!

I noticed that I could get the gravity data from sensors of iOS or Android sysem, which means the gravity components in the object coordinate system.

It seems that the Gravity data is not one of the direct output of the Fusion algorithm. So I was wondering that if it is suggested to get the Gravity just by subtracting linear acceleration from acceleration.

xioTechnologies, 2024-09-09:
Are you able to direct me to iOS and Android documentation describing this?  If it is an established feature and useful output for these systems then I would consider adding it to Fusion.

GongTen, 2024-09-10:
https://developer.apple.com/documentation/coremotion/cmdevicemotion/1616164-gravity/
https://developer.android.com/develop/sensors-and-location/sensors/sensors_motion

I sincerely hope the links above could be helpful

xioTechnologies, 2024-09-10:
Thank you.  This feature will be added to Fusion in the next few days.

GongTen, 2024-09-11:
Thanks for the quick response and the upcoming update. I'm looking forward to the feature being implemented

xioTechnologies, 2024-09-11:
This feature is now available on the branch [gravity](https://github.com/xioTechnologies/Fusion/tree/gravity).  This branch will merged into master and a new version of Fusion released in the next few days/weeks.

GongTen, 2024-09-12:
I'm glad to hear that! 

By the way, is there any plan to update the Python package, imufusion? Because I'm using the imufusion and I don't know how to build it from the source code.

xioTechnologies, 2024-09-12:
You can build and install a local version of the Python package by running `Python -m install .` in the repository root.  I suggest you change `version` in [pyproject.toml](https://github.com/xioTechnologies/Fusion/blob/main/pyproject.toml) to a district value so that your local build can be easily identified.

GongTen, 2024-09-12:
Thanks for your guidence! I have got the local build successfully by running `python setup.py install` and changing `version` in `[pyproject.toml](https://github.com/xioTechnologies/Fusion/blob/main/pyproject.toml)`. Now I can get the values of `gravity` from `imufusion.Ahrs object`

Thanks again for your quick update!

xioTechnologies, 2024-09-27:
The feature is now available in the public build, [v1.2.7](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.7).

closed 2024-09-27 by xioTechnologies