xioTechnologies/Fusion #183: [Question] Can I set an initial orientation?

msmcconnell, 2024-09-09:
I would like to initialize the algorithm from a accurate orientation quaternion coming from an INS system. Is this possible and if so what would be the correct way to do it?

Currently I see two different options
1. Set FusionAhrs.orientation and then call FusionAhrsInitialise
2. Call FusionAhrsInitialise then call FusionAhrsSetQuaternion

Which approach is correct? Or is there a better way to achieve what I want?

xioTechnologies, 2024-09-09:
`FusionAhrsInitialise` must be called the first function called for any instance of `FusionAhrs`.  This is equivalent to an class constructor.

If you want to set the initial orientation then you must skip [initialisation](https://github.com/xioTechnologies/Fusion?tab=readme-ov-file#initialisation).  The simplest way to do this is to perform a single update with a `deltaTime` value greater than [`INITIALISATION_PERIOD`](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L26).  You can then call `FusionAhrsSetQuaternion` to set the orientation.

```c
FusionAhrs ahrs;
FusionAhrsInitialise(&ahrs);

FusionAhrsUpdate(&ahrs, FUSION_VECTOR_ZERO, FUSION_VECTOR_ZERO, FUSION_VECTOR_ZERO, 4.0f); // 4 seconds is greater than INITIALISATION_PERIOD

FusionAhrsSetQuaternion(&ahrs, insQuaternion);
```

msmcconnell, 2024-09-09:
Thanks, that makes sense. I'll give it a shot.

msmcconnell, 2024-09-17:
This approach worked fine for me, so closing issue.

closed 2024-09-17 by msmcconnell