xioTechnologies/Fusion #190: Not getting output from FusionAhrsUpdate after Updating to latest version.

mjs513, 2024-09-29:
I just updated to the latest version of the Fusion and now not seeing any results, i.e.
```
    flags = FusionAhrsGetFlags(&ahrs);
    euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
    q = FusionAhrsGetQuaternion(&ahrs);
    fHeading = FusionCompassCalculateHeading(FusionConventionNed, accelerometer, magnetometer);
```
output is show all zero's.

I am using a MPU-9250 with accel/gryo axis aligned with the magnetometer axis.

Algorithm is initialized as follows:
```
    // Initialise algorithms
    FusionOffset offset;
    FusionAhrs ahrs;

    FusionOffsetInitialise(&offset, SAMPLE_RATE);
    FusionAhrsInitialise(&ahrs);
    
  // Set AHRS algorithm settings
  const FusionAhrsSettings settings = {
          .convention = FusionConventionNed,
          .gain = 0.5f, //1.5f,
          .gyroscopeRange = 2000.0f, /* replace this with actual gyroscope range in degrees/s */
          .accelerationRejection = 10.0f,
          .magneticRejection = 10.0f, //0.0f,
          .recoveryTriggerPeriod  = (uint8_t)(5.0f / SAMPLE_RATE ), /* 5 seconds */
  };
  FusionAhrsSetSettings(&ahrs, &settings);
```

Then in my fusion loop:
```
void getFusion() {
  getCalIMU();

  dt = (micros() - t_old) * 0.000001;
  t_old = micros();

    //Lets normalize magnetometer data
    //float mag_norm = sqrt(val[6]*val[6] + val[7]*val[7] + val[8]*val[8]);

    // Acquire latest sensor data
    FusionVector gyroscope = { val[3], val[4], val[5] }; // replace this with actual gyroscope data in degrees/s
    FusionVector accelerometer = { val[0], val[1], val[2] }; // replace this with actual accelerometer data in g
    FusionVector magnetometer = {  val[6], val[7], val[8] }; // replace this with actual magnetometer data in arbitrary units

    // Update gyroscope offset correction algorithm
    gyroscope = FusionOffsetUpdate(&offset, gyroscope);

    // Calculate delta time (in seconds) to account for gyroscope sample clock error
    FusionAhrsUpdate(&ahrs, gyroscope, accelerometer, magnetometer, dt);
    //FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, dt);

    flags = FusionAhrsGetFlags(&ahrs);
    euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
    q = FusionAhrsGetQuaternion(&ahrs);
    fHeading = FusionCompassCalculateHeading(FusionConventionNed, accelerometer, magnetometer);

    //if(dump > 100) {
      if(raw_values_on == 1) {
        print_float_array(val, 9);
        Serial.println();
      }

      if(serial_values_on == 1) {
        Serial.printf("Fusion (RPY): %0.1f, %0.1f, %0.1f\n", euler.angle.roll, euler.angle.pitch, euler.angle.yaw);
      }
  
      if(x_values_on == 1) {
        timestamp = micros();
        char accelgyroBuffer[100];
        sprintf(accelgyroBuffer, "%c,%llu,%0.6f,%0.6f,%0.6f,%0.6f,%0.6f,%0.6f", 'I', timestamp, val[3], val[4], val[5], val[0], val[1], val[2]);
        Serial.printf ("%s\n",accelgyroBuffer);
        sprintf(accelgyroBuffer, "%c,%llu,%0.6f,%0.6f,%0.6f", 'M', timestamp, val[6], val[7], val[8]);
        Serial.printf ("%s\n",accelgyroBuffer);
        sprintf(accelgyroBuffer, "%c,%llu,%0.6f,%0.6f,%0.6f,%0.6f", 'Q', timestamp, q.element.w, q.element.x, q.element.y, q.element.z);
        Serial.printf ("%s\n",accelgyroBuffer);
      }

      if(telem_values_on == 1) telemetryPortOut();
      //dump = 0;

}
```

Output from the MPU-9250 looks good:
[image]

When I tried with the old version it did produce results.

Any help would be appreciated.

mjs513, 2024-09-29:
Never mind think I know the issue.  Going to close this while I debug some more

mjs513, 2024-09-29:
Found one issue that I fixed but now data is kind of returning.  Bytthat I mean 'q' values and euler angles are NANs.   Flags for Initialization is 1 and timeouts are 0.

first 9 columns are the calibrated data (gyro, accel and magn),  Next are the euler angles (RPY) and next 4 are quaternions and then the flags.

```
-1.0249195099, 3.9063048363, -1.2204788923,0.1223300993,0.0566682965,-0.9995157123,-0.3305439353,-0.4308943152,0.7984790802,-NAN       ,NAN       ,NAN       ,120.6902313232,NAN       ,NAN       ,-NAN       ,-NAN       ,  1.000000,  0.000000,  0.000000-0.9638834000, 3.2349078655, -0.5490819216,0.1019417495,0.0659501702,-0.9912832975,-0.3221757412,-0.3658536673,0.8136882186,-NAN       ,NAN       ,NAN       ,127.0753326416,-NAN       ,NAN       ,-NAN       ,NAN       ,  1.000000,  0.000000,  0.000000
-0.8418111801, 3.1128358841, -0.4270097613,0.0956310704,0.0693698078,-0.9859564304,-0.3221757412,-0.3658536673,0.8136882186,-NAN       ,NAN       ,NAN       ,127.9319076538,NAN       ,NAN       ,-NAN       ,-NAN       ,  1.000000,  0.000000,  0.000000
-0.8418111801, 3.1128358841, -0.4270097613,0.0956310704,0.0693698078,-0.9859564304,-0.3221757412,-0.3658536673,0.8136882186,NAN       ,NAN       ,NAN       ,127.9319076538,NAN       ,-NAN       ,-NAN       ,NAN       ,  1.000000,  0.000000,  0.000000
-0.8418111801, 3.1128358841, -0.4270097613,0.0956310704,0.0693698078,-0.9859564304,-0.3221757412,-0.3658536673,0.8136882186,-NAN       ,-NAN       ,-NAN       ,127.9319076538,NAN       ,NAN       ,NAN       ,NAN       ,  1.000000,  0.000000,  0.000000
-0.8418111801, 3.1128358841, -0.4270097613,0.0956310704,0.0693698078,-0.9859564304,-0.3221757412,-0.3658536673,0.8136882186,-NAN       ,NAN       ,NAN       ,127.9319076538,-NAN       ,NAN       ,-NAN       ,NAN       ,  1.000000,  0.000000,  0.000000

```

xioTechnologies, 2024-09-29:
I believe that "latest version" refers to [v1.2.7](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.7).  Please confirm which version "old version" refers to.

Some comments on your code:

- `(uint8_t)(5.0f / SAMPLE_RATE)` will resolve to `0` for all values of `SAMPLE_RATE` greater than 5.  Your comment suggests that you intend `recoveryTriggerPeriod` to be 5 seconds.  For this, the expression should be `recoveryTriggerPeriod = 5 * (unsigned int) SAMPLE_RATE`.
- You have not shared variable declarations so that the type is unclear in many cases.  Expressions such as `dt = (micros() - t_old) * 0.000001;` suggest that you are not using types appropriately.
- Your calculation of `dt` incorporates unnecessary errors because you call `micros()` twice.  You should only call `micros()` per loop.

mjs513, 2024-09-30:
Morning from NY.

> I believe that "latest version" refers to [v1.2.7](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.7). Please confirm which version "old version" refers to.

Actually I downloaded the zip file directly from the repo using "CODE" option.

>Please confirm which version "old version" refers to.
Over taken by events - tried to go back to a version I had from April of 2023.  Going to try it again later today.

>For this, the expression should be recoveryTriggerPeriod = 5 * (unsigned int) SAMPLE_RATE.
Currently I am running at 100hz with `SAMPLE_RATE ` defined as 0.01 seconds.  Since the def for `recoveryTriggerPeriod` was that sample rate should be in Hz I did the diversion.

> You have not shared variable declarations so that the type is unclear in many cases. Expressions such as dt = (micros() - t_old) * 0.000001; suggest that you are not using types appropriately
> Your [...]
Yes you are right - caught that late last night.  Changed the way I did it this morning so dt is now coming out close to the sample rate. 

Even after that fix still getting nans.  Here is the whole code:
Main:
```
#include "mpu9250.h"
#include "Fusion.h"
#include <stdbool.h>
#include <stdio.h>

#include <Wire.h>
#include "Streaming.h"
#include <string>

#include "constants.h" 

/* Mpu9250 object */
bfs::Mpu9250 imu;

void gyroCalibration();

void telemetryPortOut();
void print_float_array(float *arr, int size);

bool dataRdy;

//Freeimu calibration
int cal_acc_off[] = {-37, 30, 130};
int cal_acc_scale[]  = {2047, 2060, 2065};
int cal_magn_off[]  = {99, 16, -186};
int cal_magn_scale[]  = {239, 246, 263};

// scale factors
float accelScale, gyroScale;
float magScale[3];

FusionOffset offset;
FusionAhrs ahrs;
FusionQuaternion q;
FusionEuler euler;
FusionAhrsFlags flags;

//new data available
volatile int newIMUData;
uint32_t lastUpdate, now1;

void setup() {
  /* Serial to display data */
  while(!Serial && millis() < 5000) {}
  Serial.begin(115200);
  
  // If Teensy 4.x fails print Crashreport to serial monitor
  if (CrashReport) {
      Serial.print(CrashReport);
      Serial.println("Press any key to continue");
      while (Serial.read() != -1) {
[... 342 lines cut ...]
  // Create single text parameter and print it
  snprintf(text, textLength, "%s, %s, %s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s",
           gyroxText, gyroyText, gyrozText,
           accxText, accyText, acczText,
           magxText, magyText, magzText,
           pitchText, rollText, yawText,fHeadingText,
           qwText, qxText, qyText, qzText,
           f01, f02, f03, f04, f05);
*/
    Serial.println(text);  
  //Serial.send_now();

} 

void print_float_array(float *arr, int size) {
  if (size == 0) {
    Serial.printf("[]");
  } else {
    Serial.print('[');
    for (int i = 0; i < size - 1; i++)
      Serial.printf("%f, ", arr[i]);
    Serial.printf("%f]", arr[size - 1]);
  }
}
```

mjs513, 2024-09-30:
**Fast update.  Started to get data when I commented out the following**
```
    // Update gyroscope offset correction algorithm
    gyroscope = FusionOffsetUpdate(&offset, gyroscope);
```

mjs513, 2024-10-01:
Going to close this issue out now that is working but have other questions that will post separately

closed 2024-10-01 by mjs513