xioTechnologies/Fusion #191: How to Improve Roll and Pitch Angles When Accelerometer Data is Inaccurate?

txyxy, 2024-10-17:
Firstly, Thanks for your nice work!

Our current IMU only has an accelerometer and a gyroscope, and we don't have any temperature compensation measures. We've had good results using FusionOffset to deal with gyroscope drift. However, we're facing a big issue:  when the IMU device gets hot and the algorithm starts, the initial roll and pitch angle errors become quite large, causing the world coordinate system to be skewed. Do you have any good suggestions for this?

xioTechnologies, 2024-10-17:
Your first priority should be to use a device that has good temperature stability.  For example, the [BMI088](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi088/) is 0.015 dps/K and 0.03 %/K; and the [ICM-42688-P](https://invensense.tdk.com/download-pdf/icm-42688-p-datasheet/) is 0.005 dps/K and 0.005 %/K.

Unfortunately, you cannot take all datasheet specifications at face value.  Some devices claim impressive temperature stability but are specifying a software calibrated performance where the calibration solution may be compromised in certain applications.  There may also be important performance characteristics that go largely undocumented.  For example, BMI088 has excellent vibration performance.

Your second priority should then be to perform sensor calibration at the expected operating temperature.  At the bare minimum, you should be calibrating the gyroscope offset.  Beyond that, you should calibration the accelerometer offsets and sensitivity.  This is all simple to achieve.  Further calibration, of the gyroscope sensitivity and sensor misalignments, is more difficult to achieve.

txyxy, 2024-10-23:
> Your first priority should be to use a device that has good temperature stability. For example, the [BMI088](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi088/) is 0.015 dps/K and 0.03 %/K; and the [ICM-42688-P](https://invensense.tdk.com/download-pdf/icm-42688-p-datasheet/) is 0.005 dps/K and 0.005 %/K.
> 
> Unfortunately, you cannot take all datasheet specifications at face value. Some devices claim impressive temperature stability but are specifying a software calibrated performance where the calibration solution may be compromised in certain applications. There may also be important performance characteristics that go largely undocumented. For example, BMI088 has excellent vibration performance.
> 
> Your second priority should be to perform sensor calibration at the expected operating temperature. At the bare minimum, you should be calibrating the gyroscope offset. Beyond that, you should calibration the accelerometer offsets and sensitivity. This is all simple to achieve. Further calibration, of the gyroscope sensitivity and sensor misalignments, is more difficult to achieve.

Thanks for your advices!

txyxy, 2024-10-23:
> Your first priority should be to use a device that has good temperature stability. For example, the [BMI088](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi088/) is 0.015 dps/K and [...]

Thanks for the advice, I'll give it a try.

closed 2024-10-23 by txyxy