xioTechnologies/Fusion #193: Algorithm Application for Wave Buoy IMU Data

yangleir, 2024-10-29:
First, thank you for providing this excellent open-source project! I am currently conducting research focused on wave observations using wave buoys. In my work, I analyze IMU data from the buoys to extract displacement and velocity information in the NEU (North-East-Up) coordinate system, which is essential for calculating wave parameters.

    I would like to know if your algorithm can be applied to process IMU data from wave buoys. Specifically, if it can directly derive NEU displacement or velocity from the IMU measurements, it would be extremely valuable for my research.

yangleir, 2024-10-30:
Thank you for your response.

In the context of wave observation in marine research, NEU can indeed be easily converted to NWU, ENU, or NED by simply adding a negative sign where needed. If this package can process raw IMU data and output displacement in a geographic direction similar to NWU, then I believe it could be well-suited for my research needs.

xioTechnologies, 2024-10-30:
The following answer is from the FAQ on our website.

>IMUs do not provide a direct measurement of linear velocity or position. IMUs provide inertial sensor (gyroscope and accelerometer) measurements alongside a measurement of 3D orientation relative to the Earth’s surface, often expressed as the Euler angles roll, pitch, and yaw.
>
>Our IMUs also include a magnetometer so that orientation is measured relative to magnetic north, and a sensor fusion algorithm to provide measurements of linear acceleration and earth acceleration. Earth acceleration is the [second derivative](https://en.wikipedia.org/wiki/Second_derivative) of the IMU’s position in the Earth coordinate frame.
>
>An IMU can be used to indirectly measure linear velocity and position in certain applications. However, this is only possible using specialised algorithms specific to the given application. For example, we have demonstrated how an [IMU attached to a person’s foot](https://github.com/xioTechnologies/Gait-Tracking) can be used to precisely track the 3D position of the wearer; or how [oscillatory motion](https://x-io.co.uk/oscillatory-motion-tracking-with-x-imu/) such as the heave of an ocean buoy or the respiratory motion of a patient’s chest during a medical scan can be measured using an IMU.
>
>It is important to understand that these demonstrations and algorithms are highly specialised and cannot be repurposed for generalised 3D tracking. Our IMUs are the ideal research tool for those wishing to develop tracking solutions using IMU technology. However, if you require an off-the-shelf, general-purpose, position tracking solution then you should consider alternative technologies such as GNSS or camera-based systems.

closed 2024-11-03 by yangleir