xioTechnologies/Fusion #195: [Feature request] Provide functions for generating calibration params

wheaney, 2024-11-04:
I have an IMU where I'm unable to use the magnetometer due to missing hard/soft iron calibration parameters. The fusion is still pretty solid, but it drifts horizontally over time, which I think a well calibrated magnetometer would provide the last missing piece to solve. Would it be possible for this library to provide functions for generating the magnetometer calibration values? Or any other recommendations to help solve this?

xioTechnologies, 2024-11-05:
The software mentioned in [this issue](https://github.com/xioTechnologies/Fusion/issues/29) may be what you are looking for.  Please note that Fusion now uses $m=S(u-h)$ as described [here](https://github.com/xioTechnologies/Fusion/issues/103#issuecomment-1733509838).

wheaney, 2024-11-05:
I'm specifically looking for a (preferably open source) API I can use in my own application to walk through the calibration process. My app is used by a lot of people, so this isn't just for my particular device, but trying to build out something simple that can be used by many without having to leave my app. I had a lot of trouble finding the Magneto app discussed in that issue, but I eventually tracked it down to this [old blog](https://sailboatinstruments.blogspot.com/2011/09/improved-magnetometer-calibration-part.html) and the author did provide a C-library source code link which now appears to be broken.

xioTechnologies, 2024-11-06:
I do not have an open-source solution to share.  [AN5019](https://www.nxp.com/docs/en/application-note/AN5019.pdf) is a good resource but may be an intimidating read if you don't have a background in mathematics.  I suggest you start with just hard-iron calibration as a first step.  Section [6.6.3 Offset calibration](https://x-io.co.uk/downloads/madgwick-phd-thesis.pdf) may be easier to interpret.

wheaney, 2024-11-14:
FYI and for anyone visiting this issue in the future: I was able to port some existing C# and C implementations of the Q Li Ellipsoid Fitting algo to a C library that produces the same calibration results as Magneto 1.2.

https://github.com/wheaney/magnetometer-calibrate

closed 2024-11-14 by wheaney