xioTechnologies/Fusion #196: Using BMI270/BMM150 seems to freeze algorithm for a few seconds after rapid movement?

mjs513, 2024-11-11:
Problem:
Currently I am trying to get a BMI270 (accel/gyro) and BMM150 working with the new fusion algorithm.  However, I ran into an issue where after rapid movement it seems to freeze the euler angers for a few seconds and then it recovers.

A video of the issue can be found at https://www.youtube.com/watch?v=X8hwZJyESbY.
>At about [1:15](https://www.youtube.com/watch?v=X8hwZJyESbY&t=75s) I start moving the IMU rapidly after being at rest for about 9 seconds and then put IMU back on desk (no movement).  Then it stays in that frozen position for about 30 seconds (until [1:59](https://www.youtube.com/watch?v=X8hwZJyESbY&t=119s)) and then recovers and starts updating again.  Rest of video shows is repeating the test.

To be honest I forgot to record the data file from GUI.  Here is a copy of the sketch.
[BMI270_BMM150_Fusion-241111a.zip](https://github.com/user-attachments/files/17702108/BMI270_BMM150_Fusion-241111a.zip)

xioTechnologies, 2024-11-11:
The issue appears to be with the communication of data and unrelated to Fusion.

closed 2024-11-11 by xioTechnologies

mjs513, 2024-11-11:
Not trying to be contrary but trying to figure out the comm issue you mentioned.

In this video, https://www.youtube.com/watch?v=IRFfDuMlxNU&t=12s, I let the device sit flat on the desk and then rapidly rotate it for a bit.  I then place it flat on the desk again and just rotate it around the z axis.  You can see its still responding but the angles (especially pitch and roll) are way off from flat).  I keep moving it and then towards the end of the video you see it recovers to expected orientation (flat) with just my yaw movement back to north.

You think its still a comm issue?

Here is the data files from the GUI:
[Inertial.zip](https://github.com/user-attachments/files/17703261/Inertial.zip)

xioTechnologies, 2024-11-11:
I misinterpreted the original video.  Please disregard my comment about a communication issue.  I do not understand your descriptions but I do not believe you are reporting an issue with Fusion.

mjs513, 2024-11-11:
Not sure if its fusion or something with the sensors.  I will try and explain again.

1. When I first start up Fusion and rotate about the z-axis it responds as expected.  As I rotate the sensor I see corresponding rotation about z-axis. 
[image]

2.  Now if I rotate it randomly in mulitple directions
[image]

3. and then place the sensor flat on desk and rotate it about the z-axis only it remains with what seems like the last pitch and roll of the random rotation
[image]
you can notice that the euler angle plot shows that I am rotating the sensor but that yaw and pitch never recovers to 0degrees (since sensor is flat on the desk).

4.  Now after a 10 or so seconds of it does recover pitch and roll:
[image]

not sure if its something with the sensor or fusion.

I did some further reading and looks like I should just update when gyroscope is updated and not when accel+gyro is both updated.

Will double check

mjs513, 2024-11-11:
Ok just moded the sketch to update at 100hz using only new data from the gryoscope and seeing the behavior I just post about.  Going to run some more tests but have a feel it can be closed.