xioTechnologies/Fusion #204: Q: High-g accelerometer

ib00, 2025-03-01:
What is the suggested way to use this fusion algorithm for sensors that have two accelerometers (low and high-g)
like Xio sensor?

Should we just use high-g or is there a modified algorithm that can use both accelerometers for even better
accuracy?

Thanks.

xioTechnologies, 2025-03-01:
If you are only concerned with orientation measurments (i.e. quaternion, Euler angles, rotation matrix) then the high-g accelerometer is irrelevant.  Extreme accelerations are of no benefit to the AHRS algorithm and would be ignored by the [acceleration rejection](https://github.com/xioTechnologies/Fusion?tab=readme-ov-file#acceleration-rejection) feature anyway.  The AHRS algorithm should use the most accurate accelerometer low-g measurements available.

xioTechnologies, 2025-03-02:
The [acceleration rejection](https://github.com/xioTechnologies/Fusion?tab=readme-ov-file#acceleration-rejection) is particularly effective at minimising the effects of high accelerations during impacts.  If you are observing orientation measurement errors immediately after impacts then this will likely related to the gyroscope.  Consider if your gyroscope range, sample rate, or calibration accuracy are sufficient for the dynamics you intend to measure.

ib00, 2025-03-23:
It does look like that something is wrong with the gyro signal. When stick hits the ground (with 100+ g)
the acceleration is useless for 0.1s or so, but gyro signal also seems to be impacted.

What kind of sample rate and range would you need for this type of application?

xioTechnologies, 2025-03-23:
Are you using an x-IMU3 or different hardware?  Please can you share plots, I do not understand your description.