xioTechnologies/Fusion #211: roll & pitch instability

txyxy, 2025-04-11:
Thanks for your nice work！

I’m using accelerometer and gyroscope fusion and have encountered two major issues:

1.During initialization, the roll and pitch angles sometimes have significant errors, causing the world frame’s Z-axis to appear noticeably tilted.

2.After shaking the device or performing large movements, the roll and pitch angles exhibit large errors when the device becomes stationary again. Although the system eventually stabilizes, the initial errors are quite significant.

xioTechnologies, 2025-04-11:
The algorithm outputs should be considered invalid during initialisation.  The algorithm `initialising` flag is `true` while the algorithm is initialising.

If shaking the device introduces large errors then this suggests that something about your system is insufficient to handle such motion.  This could be due to an insufficient sample rate, insufficient sensor measurement range, or poor sensor calibration.  It could also be due to an implementation error such as incorrectly configured sampling / data acquisition.

I suggest you set the `gyroscopeRange` to match your hardware and monitor the algorithm flags to see what happens during and after the problematic motion.

txyxy, 2025-05-21:
Thanks for your valuable suggestions.