xioTechnologies/Fusion #214: Yaw overshoot compare to heading

lysergicenemy, 2025-05-16:
I encountered a problem with yaw overshoot even with relatively slow turns of the board with the sensor. The heading is calculated quite accurately, although it is noisier. The graph shows the board turning by yaw by about 90 degrees. It shows that initially the yaw flies above the real value and then slowly "reaches" it. My configuration is as follows:

* ICM-20948 IMU (acc range: 8G, gyro range: +-500dps).
* according to DS for ICM-20948, mag axis Y,Z - inverted
* 50Hz sample rate
* sensor is calibrated only by offsets for the accelerometer, gyroscope and hard iron for the magnetometer
Sensor init:
`const FusionAhrsSettings settings = {

        .convention = FusionConventionEnu,
        .gain = 0.5f,
        .gyroscopeRange = 500.0f, 
        .accelerationRejection = 20.0f,
        .magneticRejection = 0.f,
        .recoveryTriggerPeriod = 5 * (1.f / p->dt), 
    };`
[image]

lysergicenemy, 2025-05-16:
Thanks for the quick reply
> My understanding is that the blue line is the yaw provided by `FusionAhrsUpdate`, and the orange line is the heading provided by `FusionCompassCalculateHeading`. Is my understanding correct?

Yes, you understand correctly.

> If you have not performed magnetic calibration then it is likely that the magnetometer is providing inaccurate measurements.

Just for hard-iron.

> For example, rotate a flat edge of the sensor 90 degrees relative to the edge of a wooden desk.

That's pretty much what I did. I concluded that the heading signal was correct based on the fact that it accurately repeated the "shape" of the motion when the board with the sensor was turned, unlike the yaw, which showed an overshoot that was not actually present in the motion.

xioTechnologies, 2025-05-16:
I'm not sure what you mean by the "shape".  It is the angle that matters.  If you physically rotate the sensor 90 degrees then you should check that the change in measured angle is 90 degrees.  The orange line appears to change by 70 degrees.  I cannot say if this matches the actual physical rotation because you have not said what the actual physical rotation is.

lysergicenemy, 2025-05-16:
> I'm not sure what you mean by the "shape"

By "shape" I mean the actual trajectory of movement when turning the sensor. For example, let's imagine that the blue graph is correct, this would mean that when turning 90 degrees I twisted it a little and returned it back. But this did not happen, I slowly and evenly turned the sensor and this trajectory corresponds to the **heading**. In this case, my question is not about accuracy (70 or 90 degrees) but about the distortion of the yaw trajectory by the Fusion.

xioTechnologies, 2025-05-17:
Fusion is attempting to combine gyroscope and magnetometer measurements.  I expect your magnetometer is poorly calibrated so that the gyroscope and magnetometer disagree.  The "overshoot" is the result of this disagreement.

During the rotation, the gyroscope affects the yaw the most and you see the yaw tracks the physical rotation, peaking at ~120°.  Once the rotation has completed, the magnetometer has a greater influence of the yaw and you see the yaw converge to the magnetometer's measurement of heading, ~70°.

You have two options:
1. If you need measurements of yaw/heading to be relative to magnetic north, then you must improve your magnetic calibration.
2. If you don't need absolute measurements of yaw/heading, then you can omit magnetometer so that the yaw/heading measurement is relative to an arbitrary starting position but liable to drift very slowly over time.

It would be very useful to know which angle was closer to the physical rotation, ~120° or ~70°.  My [comment above](https://github.com/xioTechnologies/Fusion/issues/214#issuecomment-2886972894) suggests how you could determine this.

My final suggestion is that you enable the magnetic rejection feature (e.g. `magneticRejection = 10.0f`) and then monitor the `magneticRecovery` flag.  I predict that you will see the "shape" you expect, and then 5 seconds later (determined by `recoveryTriggerPeriod`) you will see the `magneticRecovery` get set and the yaw converge the erroneous heading.

lysergicenemy, 2025-05-17:
> 1. If you need measurements of yaw/heading to be relative to magnetic north, then you must improve your magnetic calibration.

Problem solved, thanks for your help. What was done:
* reduced the gyroscope and accelerometer measurement range by 2 times.
* recalibrated offsets and hard iron offsets.

Now the yaw signal looks much better than the direction and does not have overshoot when turning the sensor to a fixed angle.

[image]

closed 2025-05-17 by lysergicenemy