xioTechnologies/Fusion #215: Yaw angle wobbling

txyxy, 2025-05-21:
Hi, thanks for your nice work again. I have a new question. I'm using accelerometer, gyroscope, and magnetometer for sensor fusion. After initialization, I placed the device completely still, but the orientation isn’t very stable — especially the yaw angle, which shows some wobbling. Here's the data:

[image]

[image]

What’s a bit strange is the magnetometer data. The figure below shows the compensated magnetometer readings.

[image]

Also, these are the parameters I'm currently using:
`  const FusionAhrsSettings settings = {`
`                    FusionConventionEnu, //convention`
`                    0.1, // gain`
`                    2000, // gyroscopeRange`
`                    6.0f, // accelerationRejection`
`                    5.0f, // magneticRejection`
`                    static_cast<unsigned int>(0.2 * m_sys_setting->imu_fps),  //recoveryTriggerPeriod`
`   };`

txyxy, 2025-05-21:
By the way, the magnetometer runs at 5 Hz — does that affect the performance of the fusion?

xioTechnologies, 2025-05-21:
The variation in the yaw is about what I would expect given the algorithm gain and the noise in your magnetometer measurements.

lovepapa, 2025-05-24:
> Btw, the magnetometer runs at 5 Hz — does that affect the performance of the fusion?

Exceuse me, did you purchse the ximu-3 product? I contact the author, but not recieved the reply. Are you also from China, I want to know how much money you cost consisting of VAT and the transport fee.

xioTechnologies, 2025-05-25:
@lovepapa Please contact us using our [website](https://x-io.co.uk/contact/), we always reply.  Please indicate in your message that you are "lovepapa" on GitHub so that we know it is you.